bigtreetech/K-Touch #1: [Issue] Automatically & Manually Adding Printers

exiom-xyz, 2024-07-29:
There are issues with adding printers,

**Automatic :**
1. Scanning for printers are inconsistent, more often than not none of my printers show up. (I have 7 klipper printers)

**Manual :**
1. Perhaps should reduce minimum characters required for IP to 7 or 8 as for example 10.0.0.2 cannot be entered as it expects at least 11 characters.

bigtreetech, 2024-07-30:
Good points. Thanks for taking the time to log these issues. 

1. Klipper printers don't advertise themselves on the network so we need to run a full scan of the whole 255 IPs on the subnet and then wait for a return ping from each IP. We can only open 10 ping channels at a time so we don't have oodles of time to wait for a response before moving on. Then we need to test for port 80 to see if it is open and then we need to submit a klipper specific web request on that port to filter out printers from other devices on the network with an open port 80. We can do a more thorough job but it would mean that the auto scan will take an age to complete. Due to the manual nature of the probing we have had to reach a balance between waiting too long on each IP (more thorough results) and providing results ASAP. We may tweak this in the future to try and give each IP a little more time to respond.
2. We will get this fixed up in the next FW release. We never caught it during testing because it seems that all of our testers had IP addresses that met the minimum length requirement.

exiom-xyz, 2024-07-31:
> 1. Klipper printers don't advertise themselves on the network so we need to run a full scan of the whole 255 IPs on the subnet and then wait for a return ping from each IP. We can only open 10 ping [...]

Would it not be simpler to probe for Moonraker at it's default port of 7125? Since Moonraker is a necessary component of a working printer whereas a WebUI is technically optional as you can add multiple printers to the same instance of Fluidd or Mainsail. 

I am assuming K-Touch interfaces with Moonraker such the same way WebUI and KlipperScreen does, so it stands to reason that it is unnecessary to check every IP for a web server and then checking it again for Klipper, but just to directly check for Klipper instead.

Fragmon, 2024-07-31:
I have the same problem. You cant add a ip with 9 digits manually.

f.e. 192.111.111.111 is not allowed. Its only possible to add 192.111.111.11

PoohWoah, 2024-08-02:
> > 1. Klipper 打印机不会在网络上播发自己，因此我们需要对子网上的整个 255 个 IP 运行全面扫描，然后等待每个 IP 的返回 ping。我们一次只能打开 10 个 ping 通道，因此我们没有大量的时间等待响应，然后再继续。然后，我们需要测试端口 80 以查看它是否打开，然后我们需要在该端口上提交特定于 klipper 的 Web 请求，以过滤掉网络上具有开放端口 80 的其他设备的打印机。我们可以做更彻底的工作，但这意味着自动扫描需要一段时间才能完成。由于探测的手动性质，我们不得不在每个 IP 上等待太久（更彻底的结果）和尽快提供结果之间取得平衡。我们将来可能会对此进行调整，以尝试给每个 IP 多一点时间来响应。
> 
> 在 Moonraker 的默认端口 7125 上探测它不是更简单吗？由于 Moonraker 是工作打印机的必要组件，而 WebUI 在技术上是可选的，因为您可以将多个打印机添加到 Fluidd 或 Mainsail 的同一实例中。
> 
> 我假设 K-Touch 与 Moonraker 的接口方式与 WebUI 和 KlipperScreen 相同，因此有理由认为没有必要检查每个 IP 的 Web 服务器，然后再次检查 Klipper，而只是直接检查 Klipper。
Hello, regarding the default "7125" port you mentioned, port changes can also be made in the configuration file moonlaker.conf. Will fixing this default port cause any inconvenience to users who have changed it? Do you have any opinion on this?
[image]

PoohWoah, 2024-08-02:
> > 我假设 K-Touch 与 Moonraker 的接口方式与 WebUI 和 KlipperScreen 相同，因此有理由认为没有必要检查每个 IP 的 Web 服务器，然后再次检查 Klipper，而只是直接检查 Klipper。
你好，对于你说的默认“7125”端口，在配置文件moonraker.conf里面也可以进行端口的更改，那么我们固定在这个默认端口是否会给那些更改了默认端口的用户造成困扰？对此你有什么看法吗？
[image]

PoohWoah, 2024-08-02:
> I am assuming K-Touch interfaces with Moonraker such the same way WebUI and KlipperScreen does, so it stands to reason that it is unnecessary to check every IP for a web server and then checking it [...]
Hello, regarding the default "7125" port you mentioned, port changes can also be made in the configuration file moonlaker.conf. Will fixing this default port cause any inconvenience to users who have changed it? Do you have any opinion on this?
[image]

PoohWoah, 2024-08-02:
> >Would it not be simpler to probe for Moonraker at it's default port of 7125? Since Moonraker is a necessary component of a working printer whereas a WebUI is technically optional as you can add multiple printers to the same instance of Fluidd or Mainsail.
I am assuming K-Touch interfaces with Moonraker such the same way WebUI and KlipperScreen does, so it stands to reason that it is unnecessary to check every IP for a web server and then checking it again for Klipper, but just to directly check for Klipper instead.

> Hello, regarding the default "7125" port you mentioned, port changes can also be made in the configuration file moonlaker.conf. Will fixing this default port cause any inconvenience to users who have changed it? Do you have any opinion on this?
[image]

exiom-xyz, 2024-08-02:
> Hello, regarding the default "7125" port you mentioned, port changes can also be made in the configuration file moonlaker.conf. Will fixing this default port cause any inconvenience to users who have changed it? Do you have any opinion on this?

Obviously I do not represent every user, and cannot speak for everybody.

As like you said and in the same vein as the IP address issues, there are a lot of different configurations out there and there indeed are people who run multiple instances of klipper on the same host (raspberry pi).

But personally, I think that for the scope of automatic detection, it would be reasonable to assume default settings as that would represent the majority of the user base and that people who are not using default settings are knowledgeable enough to manually add those instances on different ports.

So as a sidenote, it would seem without testing we have determined that the current method does not allow for adding multiple instances of Moonraker of the same IP.

bigtreetech, 2024-08-06:
> Obviously I do not represent every user, and cannot speak for everybody.
>
> As like you said and in the same vein as the IP address issues, there are a lot of different configurations out there and [...]

The current method will allow for multiple instances of MR on the same IP since all details can be specified manually and therein people can spec their desired port.

We agree on going the direct route with the MR port and will make this adjustment in the new FW.

BTTJUSTIN, 2024-09-29:
This bug was fixed in FW version V1.0.1
(Change the default port from 80 to 7125 since this is the default moonraker port.)

closed 2024-09-29 by BTTJUSTIN