bigtreetech/K-Touch #5: [Feature Request] Battery level indicator

Rushmere3D, 2024-07-31:
As above, it would be nice to have an indication of the battery level when using the K-touch on battery power.

RH3D, 2024-08-01:
In the K-Touch livestream, it was said that they run out of pins on the processor so they can't monitor and show the battery status on display because of that.

But I agree, it would be nice, although it is meant to be mostly in the dock with occasional on-battery operation.

Rushmere3D, 2024-08-01:
> In the K-Touch livestream, it was said that they run out of pins on the processor so they can't monitor and show the battery status on display because of that.
> 
> But I agree, it would be nice, although it is meant to be mostly in the dock with occasional on-battery operation.

Oh ok, wasn't able to catch the live stream. That's a bit disappointing and does bode well for future updates if pins are already fully used.

J.

closed 2024-08-01 by Rushmere3D