bigtreetech/K-Touch #6: [Issue] Chamber heater displayed as Bed

RH3D, 2024-08-01:
Chamber heater is displayed as **Bed**, so I have 2x **Bed** heater in the list. On Mobileraker or through web interface, it properly shows as a **Chamber**.
Also when trying to set the temperature for the Chamber (listed as Bed), it applies it for the other (real) Bed heater instead.

PoohWoah, 2024-08-02:
Good points. Thank you for your question. V1.0.0 only supports displaying chamber temperature. In later versions, we will add the function of setting temperature for the chamber.

RH3D, 2024-09-16:
Updating the issue as the V1.0.1 has been released and should have added the chamber heating support.
In short, it still doesn't work for me.

- **Bed** heater is now called **"heater_bed"** and has a **chamber heater icon**. Setting it's temperature sets it's temperature so it works, just looks wrong.

- **Chamber** heater is now called **"Bed"** and has a **bed heater icon**. Setting it's temperature sets the temperature for "heater_bed" so that doesn't work.

Or is there any specific way it should be set up to make it work properly with the K-Touch?

PoohWoah, 2024-09-20:
> 更新问题，因为 V1.0.1 已发布，并且应该添加了腔室加热支持。简而言之，它仍然不适合我。
> 
> * **Bed** heater 现在称为 **“heater_bed”** 并带有 **chamber heater 图标**。设置温度会设置温度，以便正常工作，只是看起来不对。
> * **腔室**加热器现在称为**“Bed”**，并带有一个 **Bed heater 图标**。设置 it's temperature 设置 “heater_bed” 的温度，因此这不起作用。
> 
> 或者应该设置任何特定方法来使其与 K-Touch 正常工作？

Greetings. Could you please send a copy about your sensor configuration? K-Touch will capture the configuration information. Currently, we divide sensors according to "extruder", "extruder" and "heater_generic" configurations.

PoohWoah, 2024-09-20:
Since many of our testers have not encountered such problems, would it be convenient for you to send a copy of your configuration file, just the part about the extruder, hot bed and chamber? This way we can quickly locate the problem.

PoohWoah, 2024-09-20:
> 更新问题，因为 V1.0.1 已发布，并且应该添加了腔室加热支持。简而言之，它仍然不适合我。
> 
> * **Bed** heater 现在称为 **“heater_bed”** 并带有 **chamber heater 图标**。设置温度会设置温度，以便正常工作，只是看起来不对。
> * **腔室**加热器现在称为**“Bed”**，并带有一个 **Bed heater 图标**。设置 it's temperature 设置 “heater_bed” 的温度，因此这不起作用。
> 
> 或者应该设置任何特定方法来使其与 K-Touch 正常工作？

For the nozzle, set the inside to [extruder], the hot bed to [heater_bed] and the chamber to [heater_generic ABS].

PoohWoah, 2024-09-20:
Would you please enter klipper's ip+ port+/printer/objects/query in your browser? Do the heaters?
Similar to: 192.168.1.135:88 /printer/objects/query? heaters.
Looking forward to your reply (obtained data)

PoohWoah, 2024-09-20:
> > > 更新问题，因为 V1.0.1 已发布，并且应该添加了腔室加热支持。简而言之，它仍然不适合我。
> > > 
> > > * **Bed** heater 现在称为 **“heater_bed”** 并带有 **chamber heater 图标**。设置温度会设置温度，以便正常工作，只是看起来不对。
> > > * **腔室**加热器现在称为**“Bed”**，并带有一个 **Bed heater 图标**。设置 it's temperature 设置 “heater_bed” 的温度，因此这不起作用。
> > > 
> > > 或者应该设置任何特定方法来使其与 K-Touch 正常工作？
> > 
> > 
> > Greetings. Could you please send a copy about your sensor configuration? K-Touch will capture the configuration information. Currently, we divide sensors according to "extruder", "heater_bed" and "heater_generic" configurations.
> 
> Hello, sure I will send you the config parameters for my heaters.
> 
> ```
> [extruder]
> step_pin: PG4
> dir_pin: !PC1
> enable_pin: !PA0
> microsteps: 16
> rotation_distance: 5.45
> nozzle_diameter: 0.400
> filament_diameter: 1.750
> heater_pin: PA2 #HE0
> sensor_type: EPCOS 100K B57560G104F
> sensor_pin: PF4 #T0
> control: pid
> pid_kp: 20.494
> pid_ki: 1.339
> pid_kd: 78.389
> min_temp: 0
> max_temp: 330
> min_extrude_temp: 170
> pressure_advance: 0.3
> pressure_advance_smooth_time: 0.03
> max_extrude_only_distance: 900.0
> max_extrude_cross_section: 60
> 
> [heater_bed]
> heater_pin: PA1
> sensor_type: Generic 3950
> sensor_pin: PF3 #TB
> min_temp: 0
> max_temp: 125
> pwm_cycle_time: 0.01
> control: pid
> pid_kp: 74.743
> pid_ki: 2.195
> pid_kd: 636.252
> 
> [heater_generic chamber]
> heater_pin: PB11
> sensor_type: EPCOS 100K B57560G104F
> sensor_pin: PF7 #T3
> min_temp: 0
> max_temp: 60
> max_power: 0.8
> control: pid
> pid_Kp: 72.832
> pid_Ki: 2.312
> pid_Kd: 573.554 #
> pwm_cycle_time: 0.1
> ```
> 
> Related to that, I also have 4 additional extruder_stepper (running a multiextruder setup) and these settings related to the chamber heater, although I don't expect them to have an impact.
> 
> ```
> [heater_fan chamber_fan] #FAN5
> pin: PD15
> heater: chamber
> heater_temp: 40
> kick_start_time: 0.5
> fan_speed: 0.8
> ```
> 
> Everything is running on BTT Octopus v1.1 I will later try renaming the to as in your example just to see. Also as mentioned previously, both Mainsail and Mobileraker see the heaters properly as they should.`[heater_generic chamber]``[heater_generic ABS]`

Would you please enter klipper's ip+ port+/printer/objects/query in your browser? heaters.
Similar to: 192.168.1.135:88 /printer/objects/query? heaters.
Looking forward to your reply (obtained data)

PoohWoah, 2024-09-20:
> > > > 更新问题，因为 V1.0.1 已发布，并且应该添加了腔室加热支持。简而言之，它仍然不适合我。
> > > > 
> > > > * **Bed** heater 现在称为 **“heater_bed”** 并带有 **chamber heater 图标**。设置温度会设置温度，以便正常工作，只是看起来不对。
> > > > * **腔室**加热器现在称为**“Bed”**，并带有一个 **Bed heater 图标**。设置 it's temperature 设置 “heater_bed” 的温度，因此这不起作用。
> > > > 
> > > > 或者应该设置任何特定方法来使其与 K-Touch 正常工作？
> > > 
> > > 
> > > 问候。您能否发送一份有关您的传感器配置的副本？K-Touch 将捕获配置信息。目前，我们根据“挤出机”、“heater_bed”和“heater_generic”配置对传感器进行划分。
> > 
> > 
> > 您好，我肯定会向您发送我的加热器的配置参数。
> > ```
> > [extruder]
> > step_pin: PG4
> > dir_pin: !PC1
> > enable_pin: !PA0
> > microsteps: 16
> > rotation_distance: 5.45
> > nozzle_diameter: 0.400
> > filament_diameter: 1.750
> > heater_pin: PA2 #HE0
> > sensor_type: EPCOS 100K B57560G104F
> > sensor_pin: PF4 #T0
> > control: pid
> > pid_kp: 20.494
> > pid_ki: 1.339
> > pid_kd: 78.389
> > min_temp: 0
> > max_temp: 330
> > min_extrude_temp: 170
> > pressure_advance: 0.3
> > pressure_advance_smooth_time: 0.03
> > max_extrude_only_distance: 900.0
> > max_extrude_cross_section: 60
> > 
> > [heater_bed]
> > heater_pin: PA1
> > sensor_type: Generic 3950
> > sensor_pin: PF3 #TB
> > min_temp: 0
> > max_temp: 125
> > pwm_cycle_time: 0.01
> > control: pid
> > pid_kp: 74.743
> > pid_ki: 2.195
> > pid_kd: 636.252
> > 
> > [heater_generic chamber]
> > heater_pin: PB11
> > sensor_type: EPCOS 100K B57560G104F
> > sensor_pin: PF7 #T3
> > min_temp: 0
> > max_temp: 60
> > max_power: 0.8
> > control: pid
> > pid_Kp: 72.832
> > pid_Ki: 2.312
> > pid_Kd: 573.554 #
> > pwm_cycle_time: 0.1
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 与此相关，我还有 4 个额外的extruder_stepper（运行多挤出机设置），这些设置与腔室加热器有关，尽管我预计它们不会产生影响。
> > ```
> > [heater_fan chamber_fan] #FAN5
> > pin: PD15
> > heater: chamber
> > heater_temp: 40
> > kick_start_time: 0.5
> > fan_speed: 0.8
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 一切都在 BTT Octopus v1.1 上运行，我稍后会尝试将重命名为 to，就像您的示例一样，只是为了看看。同样如前所述，Mainsail 和 Mobileraker 都能正确地看到加热器。`` [heater_generic chamber]``[heater_generic ABS] ``
> 
> 请在浏览器中输入 klipper 的 ip+ 端口+/printer/objects/query吗？加热器。类似于：192.168.1.135：88 /printer/objects/query？加热器。期待您的回复（获得的数据）

[image]
As is shown in the picture, he is correctly displayed on the screen.
This is the stream of data I captured:

> {
"result": {
eventtime: 23987.981266187,
"status": {
"heaters": {
"available_heaters": [
"heater_generic chamber",
"heater_bed",
"extruder"
].
"available_sensors": [
"temperature_sensor BTT-CB1",
"temperature_sensor CB1",
"heater_generic chamber",
"heater_bed",
"extruder"
].
"available_monitors": []
}
}
}

RH3D, 2024-09-20:
.......
> Would you please enter klipper's ip+ port+/printer/objects/query in your browser? heaters. Similar to: 192.168.1.135:88 /printer/objects/query? heaters. Looking forward to your reply (obtained data)

This is the data I get:

```
{
    "result": {
        "eventtime": 4342.105564411,
        "status": {
            "heaters": {
                "available_heaters": [
                    "heater_bed",
                    "heater_generic chamber",
                    "extruder"
                ],
                "available_sensors": [
                    "temperature_sensor chamber_heater",
                    "temperature_sensor raspberry_pi",
                    "temperature_sensor mcu_temp",
                    "heater_bed",
                    "heater_generic chamber",
                    "extruder"
                ],
                "available_monitors": []
            }
        }
    }
}
```
And screen looks like this:
[image]

PoohWoah, 2024-09-20:
> .......
> 
> > > Everything is running on BTT Octopus v1.1 I will later try renaming the to as in your example just to see. Also as mentioned previously, both Mainsail and Mobileraker see the heaters properly as they should.` [heater_generic chamber]`[heater_generic ABS] ``
> > 
> > 
> > Would you please enter klipper's ip+ port+/printer/objects/query in your browser? heaters. Similar to: 192.168.1.135:88 /printer/objects/query? heaters. Looking forward to your reply (obtained data)
> 
> This is the data I get:
> 
> ```
> {
>     "result": {
>         "eventtime": 4342.105564411,
>         "status": {
>             "heaters": {
>                 "available_heaters": [
>                     "heater_bed",
>                     "heater_generic chamber",
>                     "extruder"
>                 ],
>                 "available_sensors": [
>                     "temperature_sensor chamber_heater",
>                     "temperature_sensor raspberry_pi",
>                     "temperature_sensor mcu_temp",
>                     "heater_bed",
>                     "heater_generic chamber",
>                     "extruder"
>                 ],
>                 "available_monitors": []
>             }
>         }
>     }
> }
> ```
> 
> And screen looks like this: ![IMG_20240920_110956045_HDR](https://private-user-images.githubusercontent.com/127200179/369342749-7fe03718-3736-4726-b0c0-b684d0277ab6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY4MjQwNjEsIm5iZiI6MTcyNjgyMzc2MSwicGF0aCI6Ii8xMjcyMDAxNzkvMzY5MzQyNzQ5LTdmZTAzNzE4LTM3MzYtNDcyNi1iMGMwLWI2ODRkMDI3N2FiNi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMFQwOTE2MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iYTlkZmM5MTAxMzA2ZTJjNjBkZmUxY2ExZjI5YTdhZDY4OTllNmI0ZmU2NzlkNmE2NzVjMWYzNTIwYWQyZjU5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.d-rpMXdW4id-REFBwVrUoWbnt6zjwFjw369IGsqDpGU)

Ok, I got it. It's a real problem. We'll fix it as soon as possible. A smaller version, V1.0.1.1, will be updated tonight. Hope you can update and try again.

PoohWoah, 2024-09-20:
In addition, you can change the order of the configuration files to [extruder], [heater_bed], [heater_generic chamber]. I think the problem will be solved.

> .......
>
>
> This is the data I get:
>
> ```
> {
>     "result": {
>         "eventtime": 4342.105564411,
>         "status": {
>             "heaters": {
>                 "available_heaters": [ [...]

In addition, you can change the order of the configuration files to [extruder], [heater_bed], [heater_generic chamber]. I think the problem will be solved.

RH3D, 2024-09-20:
...
> 
> Ok, I got it. It's a real problem. We'll fix it as soon as possible. A smaller version, V1.0.1.1, will be updated tonight. Hope you can update and try again.

Perfect, thank you, I will update and test it once it is released, but it is not crucial for me at the moment.

PS my config is in the order you mentioned - [extruder], [heater_bed], [heater_generic chamber], but I might try ordering them differently or trying to rename the chamber after I finish the print, just to see if anything solves it.

PoohWoah, 2024-09-20:
> ...
> 
> > 好的，我明白了。这是一个真正的问题。我们会尽快修复它。较小的版本 V1.0.1.1 将于今晚更新。希望您可以更新并重试。
> 
> 完美，谢谢，一旦发布，我会更新和测试它，但目前对我来说并不重要。
> 
> PS 我的配置按照您提到的顺序 - [挤出机]、[heater_bed]、[heater_generic腔室]，但我可能会尝试以不同的方式排序它们或在完成打印后尝试重命名腔室，只是为了看看是否有任何方法可以解决它。

If this sort will solve the problem, you can test it after printing, thank you again for your feedback. Thank you!

RH3D, 2024-09-21:
> > ...
> > > 好的，我明白了。这是一个真正的问题。我们会尽快修复它。较小的版本 V1.0.1.1 将于今晚更新。希望您可以更新并重试。
> > 
> > 
> > 完美，谢谢，一旦发布，我会更新和测试它，但目前对我来说并不重要。
> > PS 我的配置按照您提到的顺序 - [挤出机]、[heater_bed]、[heater_generic腔室]，但我可能会尝试以不同的方式排序它们或在完成打印后尝试重命名腔室，只是为了看看是否有任何方法可以解决它。
> 
> If this sort will solve the problem, you can test it after printing, thank you again for your feedback. Have a nice day.

So I tried changing the order to [extruder], [heater_generic chamber], [heater_bed] and now it shows and works the way it should.
Thank you! Have a nice day too.

PoohWoah, 2024-09-26:
> Updating the issue as the V1.0.1 has been released and should have added the chamber heating support. In short, it still doesn't work for me.
>
> * **Bed** heater is now called **"heater_bed"** and has [...]

Send one's regards to
The new firmware version has been updated to fix this problem, we have updated the sorting algorithm, and now you should be able to configure it at will, please download and experience it, looking forward to your feedback. thank you.

RH3D, 2024-09-26:
> Send one's regards to The new firmware version has been updated to fix this problem, we have updated the sorting algorithm, and now you should be able to configure it at will, please download and [...]

Perfect! Short test of rearranging heaters in the printer.cfg still gives the same result and everything seems to be working properly now.
Thank you for the update/fix!

Closing the issue as it's been solved with v1.0.1.1 FW update.

closed 2024-09-26 by RH3D

PoohWoah, 2024-09-26:
> > > 更新问题，因为 V1.0.1 已发布，并且应该添加了腔室加热支持。简而言之，它仍然不适合我。
> > > 
> > > * **Bed** heater 现在称为 **“heater_bed”** 并带有 **chamber heater 图标**。设置温度会设置温度，以便正常工作，只是看起来不对。
> > > * **腔室**加热器现在称为**“Bed”**，并带有一个 **Bed heater 图标**。设置 it's temperature 设置 “heater_bed” 的温度，因此这不起作用。
> > > 
> > > 或者应该设置任何特定方法来使其与 K-Touch 正常工作？
> > 
> > 
> > 送上问候新的固件版本已经更新解决了这个问题，我们更新了排序算法，现在你应该可以随意配置了，请下载体验，期待你的反馈。谢谢。
> 
> 完善！在 printer.cfg 中对重新排列加热器进行简短测试仍然给出了相同的结果，现在一切似乎都正常工作。感谢您的更新/修复！
> 
> 关闭问题，因为它已通过 v1.0.1.1 固件更新解决。

Thank you for your reply, we will close the question later.

PoohWoah, 2024-09-26:
> > > > 更新问题，因为 V1.0.1 已发布，并且应该添加了腔室加热支持。简而言之，它仍然不适合我。
> > > > 
> > > > * **Bed** heater 现在称为 **“heater_bed”** 并带有 **chamber heater 图标**。设置温度会设置温度，以便正常工作，只是看起来不对。
> > > > * **腔室**加热器现在称为**“Bed”**，并带有一个 **Bed heater 图标**。设置 it's temperature 设置 “heater_bed” 的温度，因此这不起作用。
> > > > 
> > > > 或者应该设置任何特定方法来使其与 K-Touch 正常工作？
> > > 
> > > 
> > > 送上问候新的固件版本已经更新解决了这个问题，我们更新了排序算法，现在你应该可以随意配置了，请下载体验，期待你的反馈。谢谢。
> > 
> > 
> > 完善！在 printer.cfg 中对重新排列加热器进行简短测试仍然给出了相同的结果，现在一切似乎都正常工作。感谢您的更新/修复！
> > 关闭问题，因为它已通过 v1.0.1.1 固件更新解决。
> 
> Thank you for your reply, I'm glad the problem has been resolved.