bigtreetech/K-Touch #10: [issue] Cannot enter Short IP's or Long IP's

mon5termatt, 2024-08-03:
The Ktouch does not allow Short IP addresses 

Examples 

❌10.0.0.224
❌10.0.0.10
(too short) 
✔010.0.0.224

❌192.168.123.123
(too many)

mon5termatt, 2024-08-03:
mentioned in #1 but that also has other issues in that issue. Making this one for ease of tracking.

PoohWoah, 2024-08-03:
Good points. Thanks for taking the time to log these issues.
We will get this fixed up in the next FW release. We never caught it during testing because it seems that all of our testers had IP addresses that met the minimum length requirement.

mon5termatt, 2024-08-03:
Some simple regex would be a lot better.

https://stackoverflow.com/questions/5284147/validating-ipv4-addresses-with-regexp

BTTJUSTIN, 2024-09-29:
This bug was fixed in FW version V1.0.1

closed 2024-09-29 by BTTJUSTIN