bigtreetech/K-Touch #9: Loading Large Thumbnails Makes UI unresponsive.

mon5termatt, 2024-08-03:
https://youtu.be/QYWDmYZsOlE

mon5termatt, 2024-08-03:
Thumbnails seem to cover the stats of the files. After a minute or so, the screen decides to scroll randomly and makes the UI unresponsive.

PoohWoah, 2024-08-03:
Hello, thank you for raising the question. It is best to maintain a file thumbnail resolution of around 100x100 when reading on K-Touch. In the next version, K-Touch will adaptively adjust the size of the image based on the thumbnail resolution (provided that the resolution is below 100x100). This way, you can view higher definition thumbnails.

PoohWoah, 2024-08-03:
Hello, thank you for raising the question. Please maintain the file thumbnail resolution for reading on K-Touch at around 100x100. If the file thumbnail resolution is too high, it will cause K-Touch resources to run out. So, in the current version, please set the thumbnail to around 100x100 when slicing to maintain K-Touch's normal reading of file thumbnails. In the next version of K-Touch, the image will be resized adaptively based on the resolution of the thumbnail. This way, you can view higher definition thumbnails.

PoohWoah, 2024-08-03:
Hello, thank you for raising the question. Please maintain the file thumbnail resolution for reading on K-Touch at around 32x32. Currently, K-Touch scales based on this size, so if the file thumbnail resolution is too high, it will cause K-Touch to exhaust its resources and exceed its display range. So, in the current version, please set the thumbnail to around 32x32 when slicing to maintain K-Touch's normal reading of file thumbnails. In the next version of K-Touch, the image will be resized adaptively based on the resolution of the thumbnail. This way, you can view higher definition and more consistent sized thumbnails.

mon5termatt, 2024-08-04:
32x32??? What are you going to even be able to make out at 32x32. That's not a thumbnail that's pixel art....

PoohWoah, 2024-08-04:
> 32x32???你甚至能在 32x32 上看出什么。那不是像素艺术的缩略图......它是蚂蚁的缩略图！

More high-definition thumbnails will be displayed in the future, as defined in the current version. For the current 32x32 resolution thumbnail, we will enlarge it for you to view, but what you actually see is not 32x32.

PoohWoah, 2024-08-04:
> 32x32???你甚至能在 32x32 上看出什么。那不是像素艺术的缩略图......它是蚂蚁的缩略图！

For the current 32x32 resolution thumbnail, we will enlarge it for you to view, but what you actually see is not 32x32. You can actually check it out.

PoohWoah, 2024-08-04:
> 32x32???你甚至能在 32x32 上看出什么。那不是像素艺术的缩略图......它是蚂蚁的缩略图！

我们会使用别的方法让缩略图变大，这样做的目的只是为之后K-Touch能够读取到更多的缩略图而作的准备。比如我现在读取到的就是设置好分辨率的图片
[image]
可以看到它已经放大了，可以让我们清晰的看到。

PoohWoah, 2024-08-04:
> 32x32??? What are you going to even be able to make out at 32x32. That's not a thumbnail that's pixel art.... Its a thumbnail for ants!

We will use other methods to make the thumbnail larger, but the purpose of doing so now is only to prepare for K-Touch to be able to read more thumbnails in the future. For example, what I am currently reading is an image with a set resolution.
[image]
可以看到它已经放大了，可以让我们清晰的看到。

PoohWoah, 2024-08-05:
Hello. Also, can you provide me with a GCode file?

exiom-xyz, 2024-08-05:
> 32x32??? What are you going to even be able to make out at 32x32. That's not a thumbnail that's pixel art.... Its a thumbnail for ants!

if i recall correctly... the K-Touch/PandaTouch is based on the ESP32 so you need to keep in mind of it's performance capabilities... it is not like KlipperScreen where it runs off your raspberry pi or other SBC... for that you need something like the BTT Pad 7

mon5termatt, 2024-08-05:
i understand that its running on a smaller cpu (heck i use esp32 boards all the time). And it doesn't bother me all that much, but the text not being on top of the thumbnail should be an important bug.

mon5termatt, 2024-08-05:
or allow fewer items on screen with larger thumbnails,

PoohWoah, 2024-08-05:
> 或者允许屏幕上使用较大的缩略图减少项目，

Yes, the next version will become an adaptive thumbnail. Currently, we only capture the 32X32 thumbnail that it defaults to transmitting to us. Our testers have not encountered the situation you mentioned. Can you provide me with a gcode file and we will check if there are any other issues.

mon5termatt, 2024-08-05:
i cannot provide the same file as it has copyright to it, and i cannot share it, however i will provide a benchy using my settings.

mon5termatt, 2024-08-05:
[Benchy400.zip](https://github.com/user-attachments/files/16492886/Benchy400.zip)

mon5termatt, 2024-08-05:
[image]

PoohWoah, 2024-08-05:
> 我无法提供相同的文件，因为它具有版权，并且我无法共享它，但是我将使用我的设置提供一个长凳。

It's okay, thank you.

BTTJUSTIN, 2024-09-29:
This bug was fixed in FW version V1.0.1

closed 2024-09-29 by BTTJUSTIN