bigtreetech/K-Touch #13: Battery status indicator

Spolfox, 2024-08-07:
Make a Battery Status Indicator. So that when you use it, you do not allow a full discharge. And it will also be visible when it is charged and you can remove it from the printer and go with it.

Spolfox, 2024-08-07:
@bigtreetech

@K-touch

pogletree, 2024-08-07:
Probably the easiest is to add a LED charge indicator in the next version PC board and case design. Either with a "light-pipe" or LED-in-case-hole. 
Software solutions would require that the LCD be "hot" during the charge for charging status. An "on" battery status would be easy to display IF the battery voltage is monitored already.

exiom-xyz, 2024-08-08:
In the K-Touch livestream, it was said that they have run out of pins on the processor so they can't monitor and show the battery status on display because of that.

bigtreetech, 2024-08-08:
Unfortunately this version of hardware cannot support that because the TFT occupies so many pins on the MCU that there is literally not a single one remaining. We are looking at alternative ways to implement this though.

closed 2024-08-08 by bigtreetech

Spolfox, 2024-08-08:
I'm not talking about a physical indicator or LED.
I am talking about the indicator at the software level, as in smartphones, watches, blood pressure monitors, everything that has batteries.
and other equipment