bigtreetech/K-Touch #22: Need a way to invert Z+ and Z-

nicwilson58, 2024-08-11:
Due top my Klipper setup,  Z+ and Z- are inverted,  on KlipperScreen it was easy to invert them by setting this in the Klipperscreen config,   and also on Fluidd they work the correct way round,  but on Ktouch they are inverted,  so Z+ goes down,  and Z- goes up so this needs inverting,  by aadding a setting to do so or similar as other interfaces have.

Rowan2233, 2024-08-12:
Thank you bro, we will update this feature in v1.1.0

nicwilson58, 2024-09-16:
> @nicwilson58<https://github.com/nicwilson58> I wonder if this should really be needed in any control screen. As you say it is due to your Klipper setup, not just K-Touch specific.
> Doesn't inverting the direction of the Z axis in your printer config file work? (putting "!" before the dir_pin PIN name) It would be much simpler.

Hi,

No the printer already goes the right way,  but the button the screen of the k-touch do the opposite.  This is a common issue  and klipperscreen provides a way to invert the screen buttons function.

Nic