bigtreetech/K-Touch #23: Creality K1 c and K1 max and K touch

Gismo6, 2024-08-16:
Have Creality K1 c and K1 max. The K touch cannot find these printers. IP address does not work because my IP address is longer 15 numbers, I can only enter 14.

bistory, 2024-08-19:
Did you root your K1's and installed the a more recent version of Moonraker ?

Gismo6, 2024-08-21:
no

PoohWoah, 2024-08-22:
Okay, thank you for your question. The IP input issue will be fixed in the next version. Please refer to the details #1

bistory, 2024-08-22:
> no

You have to, even when the ip issue will be fixer.

Andy3dprinter, 2024-08-26:
> Have Creality K1 c and K1 max. The K touch cannot find these printers. IP address does not work because my IP address is longer 15 numbers, I can only enter 14.

I am running mine on a K1 however the others are correct. I rooted mine a long time ago and am using the script from Guilouz and it's been great. I rooted and installed full Klipper on the second day I bought it. It is worth it and you will get to learn and understand Klipper and things like Mainsail or Fluidd. :)

BTTJUSTIN, 2024-09-29:
This bug was fixed in FW version V1.0.1

closed 2024-09-29 by BTTJUSTIN