bigtreetech/K-Touch #29: New update won’t load new filament

mendeste, 2024-09-15:
Update to new firmware  Now won't load filament  Screen pop up set temp to 200 click ok and nothing happens

PoohWoah, 2024-09-18:
Hello, do you mean he can't heat the nozzle?

mendeste, 2024-09-18:
No I mean
The nozzle heats to 200
Then you tap extrude and the same message comes up
If I revert back to previous firmware no problem

Regards

PoohWoah, 2024-09-18:
Ok, I see what you mean. We will fix this problem as soon as possible. Thank you.

mendeste, 2024-09-24:
Hi PoohWoah 
when will this problem be fixed
Regards

PoohWoah, 2024-09-25:
> 嗨，PoohWoah，这个问题什么时候能解决 问候
Hello, for this problem, we will issue a V1.0.1.1 to fix it, the new firmware will be pushed to git this week, please pay attention. Have a nice day.

PoohWoah, 2024-09-26:
> Hi PoohWoah when will this problem be fixed Regards

Send one's regards to
The new firmware version has been updated, and this problem has been fixed. Please download it and try it. Look forward to your feedback. thank you.

mendeste, 2024-09-26:
Hi PoohWoah

Yes all working great

Thank you

PoohWoah, 2024-09-27:
> 嗨 PoohWoah 是的，一切都运行良好 谢谢
> […](#)

Thank you for your reply. Glad to solve your problem.

closed 2025-01-25 by BTTJUSTIN