bigtreetech/K-Touch #33: Only shows black screen

jefferson-sandals, 2024-09-19:
K-Touch is only showing a blank black screen whenever it powers on.

PoohWoah, 2024-09-20:
Hello, can you describe it more clearly?

jefferson-sandals, 2024-09-20:
Whether powered by battery or 5V, it turns on, but the screen is blank.
Nothing is displayed.

PoohWoah, 2024-09-20:
> 无论是用电池供电还是 5V 供电，它都会打开，但屏幕是空白的。 不显示任何内容。
> […](#)

You can download the firmware again using the flash_download_tool tool according to the last section of the manual.

jefferson-sandals, 2024-09-20:
> 无论是用电池供电还是 5V 供电，它都会打开，但屏幕是空白的。 不显示任何内容。
> … <#m_-249725829283240626_>
> On Fri, Sep 20, 2024 at 00:18 PoohWoah *@*.*> wrote: Hello, can you

Do I use the included dock usb-c? Can I connect through the usb drive usb
port?

PoohWoah, 2024-09-20:
You can connect the computer directly using USB, we used USB to serial port, and then you can choose the COM port and other Settings according to the instructions.

jefferson-sandals, 2024-09-20:
USB from PC to serial port (the I2C port on the back)? Where do I find a
cable for that??

PoohWoah, 2024-09-23:
> 从 PC 的 USB 连接到串行端口（背面的 I2C 端口）？我在哪里可以找到 电缆？？
> […](#)

Hello, the charging port is a serial port, merged together.

closed 2025-01-25 by BTTJUSTIN