bigtreetech/K-Touch #36: [ISSUE] Extrude / retract requires 200°C at hotend

RH3D, 2024-09-21:
Both extrude and retract buttons require exactly 200°C on hotend [extruder].
I am not sure if this was in the previous FW but I didn't notice it.

I think this should preferably follow the user set `min_extrude_temp` value in printer.cfg or just leave the temperature check on the Klipper itself.

This way I can't extrude filaments that require higher temps.
[image]

PoohWoah, 2024-09-23:
> 挤出和缩回按钮在热端 [挤出机] 上都需要 200°C。我不确定这是否在之前的 FW 中，但我没有注意到。
> 
> 我认为这最好遵循 printer.cfg 中的用户设置值，或者将温度检查留在 Klipper 本身上。`min_extrude_temp`
> 
> 这样我就无法挤出需要更高温度的细丝。 ![IMG_20240920_170233550_HDR-01](https://private-user-images.githubusercontent.com/127200179/369618227-adcab1ff-0c53-4add-88fd-94f311e1e21b.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcwNTM5MTMsIm5iZiI6MTcyNzA1MzYxMywicGF0aCI6Ii8xMjcyMDAxNzkvMzY5NjE4MjI3LWFkY2FiMWZmLTBjNTMtNGFkZC04OGZkLTk0ZjMxMWUxZTIxYi5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MjNUMDEwNjUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDJlYmUyN2QyNWJmMzQ5Mjk1YmNkMGI4NDE4NWM3ZTRjM2I4ODRiYTI4ZWEyNmE3OThhNjU2MTE3MGJlOTNlMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.0aRtSFP2eZX3GwKkmGq15i0aewyGTQiF0wjWX-KYeho)

Hi, thank you for your feedback, we will consider this issue in a later version. thank you.

RH3D, 2024-09-26:
> Hi, thank you for your feedback, we will consider this issue in a later version. thank you.

Thank you for the fix, it now works properly with any temperature higher than 200°C.

I still think it would be better to follow the printer.cfg `min_extrude_temp` value set by user, but it is definitely not necessary and the way it works now is OK.

Closing the issue as it is now solved with v1.0.1.1 FW update.

closed 2024-09-26 by RH3D

PoohWoah, 2024-09-26:
> > > 挤出和缩回按钮在热端 [挤出机] 上都需要 200°C。我不确定这是否在之前的 FW 中，但我没有注意到。 我认为这最好遵循 printer.cfg 中的用户设置值，或者将温度检查留在 Klipper 本身上。 这样我就无法挤出需要更高温度的细丝。`min_extrude_temp`![IMG_20240920_170233550_HDR-01](https://private-user-images.githubusercontent.com/127200179/369618227-adcab1ff-0c53-4add-88fd-94f311e1e21b.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcwNTM5MTMsIm5iZiI6MTcyNzA1MzYxMywicGF0aCI6Ii8xMjcyMDAxNzkvMzY5NjE4MjI3LWFkY2FiMWZmLTBjNTMtNGFkZC04OGZkLTk0ZjMxMWUxZTIxYi5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MjNUMDEwNjUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDJlYmUyN2QyNWJmMzQ5Mjk1YmNkMGI4NDE4NWM3ZTRjM2I4ODRiYTI4ZWEyNmE3OThhNjU2MTE3MGJlOTNlMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.0aRtSFP2eZX3GwKkmGq15i0aewyGTQiF0wjWX-KYeho)
> > 
> > 
> > 您好，感谢您的反馈，我们将在以后的版本中考虑此问题。谢谢。
> 
> 感谢您的修复，它现在可以在高于 200°C 的任何温度下正常工作。
> 
> 我仍然认为遵循用户设置的 printer.cfg 值会更好，但这绝对不是必需的，现在它的工作方式还可以。`min_extrude_temp`
> 
> 关闭该问题，因为它现在已通过 v1.0.1.1 固件更新解决。

Thank you for your reply. We will consider the problem later.