bigtreetech/K-Touch #49: 闪退重启1.1版本

nbkls456, 2024-11-27:
kt更新1.1版本，遇到以下问题：1点击pid校准开始就会重启。2点击其他按键也有几率重启，3连接打印机不稳定，有的时候得切换2台打印机重新旋转打印机才会连接正常。  回转了1.01版本然后pid校准点击没效果

PoohWoah, 2024-11-27:
你好，请问第二点点击其他按键也有几率重启是什么意思，可以说详细点吗
第三点里面，这个连接打印机是根据网络情况来获取的

PoohWoah, 2024-11-27:
> kt更新1.1版本，遇到以下问题：1点击pid校准开始就会重启。2点击其他按键也有几率重启，3连接打印机不稳定，有的时候得切换2台打印机重新旋转打印机才会连接正常。 回转了1.01版本然后pid校准点击没效果

问题一，在我们测试的时候没有发现，之后我们会在下个版本进行解决
问题二，我们更新了大版本后，您应该重置k-touch，不然就会出现此情况，因为数据存储的地方发生了改变，
问题三，这个问题是由于klipper那边的通信是http，在网络不好的时候就会连接不上打印机。

PoohWoah, 2024-11-27:
> kt更新1.1版本，遇到以下问题：1点击pid校准开始就会重启。2点击其他按键也有几率重启，3连接打印机不稳定，有的时候得切换2台打印机重新旋转打印机才会连接正常。 回转了1.01版本然后pid校准点击没效果

在进行更新时可以去看看我们的官方wiki，里面有更新时需要进行的操作，这样可以减少你出现问题时候的疑惑
https://bttwiki.com/zh/Klipper%20Touch.html#v101

nbkls456, 2024-11-30:
我重按教程更新一次还是有这个问题

PoohWoah, 2024-12-02:
> 我重按教程更新一次还是有这个问题

PID的问题吗  还是别的  先重置后在操作

PoohWoah, 2024-12-03:
> 重新OTA了，打开pid点击开始还是会重启

这个之后会修复

PoohWoah, 2024-12-03:
> 重新OTA了，打开pid点击开始还是会重启

你好，请看上面问题一我的回答

nbkls456, 2024-12-04:
> > 重新OTA了，打开pid点击开始还是会重启
> 
> 你好，请看上面问题一我的回答

都有，重置后打开某些按键一样会有重启的情况，过段时间操作也会重启，pid是百分百重启

PoohWoah, 2024-12-05:
> > > 重新OTA了，打开pid点击开始还是会重启
> > 
> > 
> > 你好，请看上面问题一我的回答
> 
> 都有，重置后打开某些按键一样会有重启的情况，过段时间操作也会重启，pid是百分百重启

某些按键是哪些按键，可以给个视频么，PID的我已经知道是怎么回事了

nbkls456, 2024-12-06:
没有视频了，不过现在没遇到其他黑屏了，就剩pid了