bigtreetech/K-Touch #51: Are we able to disable widgets?

Egosumumbravir, 2024-12-07:
Adding a Panda-PWR to my Klipper machine so have updated the firmware to v1.1.0 and discovered the new widgets.

Are we able to disable widgets?  Pressing, holding and dragging to reorder results in a reboot that doesn't appear to save the reorder changes despite a "settings saved successfully" message. The cycle pages toggle does appear to work though.

PoohWoah, 2024-12-09:
> 将 Panda-PWR 添加到我的 Klipper 机器上，因此已将固件更新到 v1.1.0 并发现了新的小部件。
> 
> 我们能禁用小部件吗？按住并拖动以重新排序会导致重新启动，尽管显示“settings saved successfully”消息，但似乎不会保存重新排序的更改。不过，循环页面切换似乎确实有效。

Hello, you can turn off the automatic polling widget by clicking the switch below the widget.
For the problem of restart, we will update it in the next minor version, please understand.

Egosumumbravir, 2024-12-09:
Yes, automatic polling does work as expected.

Would be a nice feature to disable unwanted/unnecessary panels too.