bigtreetech/K-Touch #53: Flash from Panda to K-Touch on MAC

wsturner88, 2024-12-15:
In my attempts to update firmware, I was not aware there is a different firmware between Panda and K-Touch.   I used the Macro method to Flash the Panda.   Is there a firmware I can use to Flash the K-touch software over the Panda Software?

Pneumanifest, 2024-12-16:
https://github.com/bigtreetech/K-Touch

wsturner88, 2024-12-16:
Even when you want to switch from Panda to K-touch?   Wow.. I thought I read that was not possible.  I will give it a try!

PoohWoah, 2024-12-20:
> Even when you want to switch from Panda to K-touch? Wow.. I thought I read that was not possible. I will give it a try!即使您想从 Panda 切换到 K-touch？哇..我以为我读到这是不可能的。我会尝试一下！

Panda Touch and K-Touch can be universal, when you update, you need to record the Panda Touch data, and then restore the factory Settings, in the OTA update firmware. Because the data storage methods of the two firmware are different, the display may be confused if the switch is made at will.

BTTJUSTIN, 2025-02-27:
The post without any information updates for more than three weeks.

closed 2025-02-27 by BTTJUSTIN