bigtreetech/K-Touch #54: Add instructions for flashing direct via esptool

vvuk, 2024-12-21:
In case anyone has issues with the built-in updating, you can flash directly with `esptool.py`:

```
esptool.py write_flash 0x0 K-Touch_v1.1.0_bootloader.bin 0x8000 K-Touch_v1.1.0_partition.bin 0x10000 K-Touch_v1.1.0_firmware.bin 
```

Would be nice to have this added to instructions somewhere.

thiagolocatelli, 2025-01-11:
@vvuk was that on a Windows or Mac machine? I am trying the same command on my mac and it just doesnt wok

```
[thiago@multiverse ~/Downloads/K-Touch/v1.1.0 (master)] esptool.py write_flash     0x0 K-Touch_v1.1.0_bootloader.bin     0x8000 K-Touch_v1.1.0_partition.bin     0x10000 K-Touch_v1.1.0_firmware.bin     0x910000 K-Touch_v1.1.0_product.img
esptool.py v4.8.1
Found 1 serial ports
Serial port /dev/cu.debug-console
Connecting......................................
/dev/cu.debug-console failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.
```

vvuk, 2025-01-11:
On mac, esptool may not find the right port by itself -- you'll have to specify it explicitly.

thiagolocatelli, 2025-01-11:
yeah, I am on a Mac M2. Do you recall remember port you specified?

vvuk, 2025-01-11:
My port won't be the same as yours -- look in /dev/, you're looking for something that looks like `/dev/tty.usbserial-0001`. An easy way would be to unplug the device, do a `ls /dev/tty.*`, then plug it in and do an ls again -- it'll be whatever device showed up.

thiagolocatelli, 2025-01-11:
and you didnt have to install any drive or anything? I have tried all that and no luck, I dont get anything useful in /dev/tty.*