bigtreetech/K-Touch #55: Thumbnails from PrusaSlicer and SuperSlicer not recognised

Hr-Ravn, 2024-12-27:
K-touch with Firmware 1.1.0 installed is not recognising the thumbnails wether it's set for 100x100 PNG, 200x200 PNG or 300x300 PNG as recommended, when slicing gcode from PrusaSlicer, PrusaSlicer BOSS (https://github.com/mjonuschat/PrusaSlicer/releases/tag/version_2.9.0%2Bboss) or SuperSlicer.

I've tried out OrcaSlicer, which seems to work. 

Could the firmware be configured to work with the other slicers as well? 

I've uploaded a PrusaSlicer BOSS gcode file and a OrcaSlicer file to show you the difference/similarities. 
[Gcodes.zip](https://github.com/user-attachments/files/18260012/Gcodes.zip)

PoohWoah, 2025-01-02:
> 从 PrusaSlicer、PrusaSlicer BOSS （https://github.com/mjonuschat/PrusaSlicer/releases/tag/version_2.9.0%2Bboss） 或 SuperSlicer 对 gcode 进行切片时，安装了固件 1.1.0 的 K-touch 无法识别缩略图，无论它是否按照建议设置为 100x100 PNG、200x200 PNG 或 300x300 PNG。
> 
> 我已经试用了 OrcaSlicer，它似乎有效。
> 
> 固件也可以配置为与其他切片器一起使用吗？
> 
> 我已经上传了一个 PrusaSlicer BOSS gcode 文件和一个 OrcaSlicer 文件来向您展示差异/相似之处。 [Gcodes.zip](https://github.com/user-attachments/files/18260012/Gcodes.zip)

Hello, when slicing we can select the script to generate the desired thumbnail size, and K-Touch reads the first thumbnail size by default.
We haven't tested OrcaSlicer yet, but we will test it later.

Hr-Ravn, 2025-01-02:
> Hello, when slicing we can select the script to generate the desired thumbnail size, and K-Touch reads the first thumbnail size by default. We haven't tested OrcaSlicer yet, but we will test it later.

Hi PoohWoah

Which script are you using for this? 
I'm just setting the Thumbnail size : 200X200/PNG in the printer tab, under thumbnail settings.

Hr-Ravn, 2025-01-04:
> Hi PoohWoah
>
> Which script are you using for this? I'm just setting the Thumbnail size : 200X200/PNG in the printer tab, under thumbnail settings.

PoohWoah, 2025-02-07:
> > > 从 PrusaSlicer、PrusaSlicer BOSS （[https://github.com/mjonuschat/PrusaSlicer/releases/tag/version_2.9.0%2Bboss）](https://github.com/mjonuschat/PrusaSlicer/releases/tag/version_2.9.0%2Bboss%EF%BC%89) 或 SuperSlicer 对 gcode 进行切片时，安装了固件 1.1.0 的 K-touch 无法识别缩略图，无论它是否按照建议设置为 100x100 PNG、200x200 PNG 或 300x300 PNG。
> > > 我已经试用了 OrcaSlicer，它似乎有效。
> > > 固件也可以配置为与其他切片器一起使用吗？
> > > 我已经上传了一个 PrusaSlicer BOSS gcode 文件和一个 OrcaSlicer 文件来向您展示差异/相似之处。 [Gcodes.zip](https://github.com/user-attachments/files/18260012/Gcodes.zip)
> > 
> > 
> > Hello, when slicing we can select the script to generate the desired thumbnail size, and K-Touch reads the first thumbnail size by default. We haven't tested OrcaSlicer yet, but we will test it later.您好，切片时，我们可以选择脚本来生成所需的缩略图大小，并且K-Touch默认情况下读取第一个缩略图大小。我们尚未测试orcaslicer，但稍后将测试。
> 
> Hi PoohWoah  嗨，Poohwoah
> 
> Which script are you using for this?您正在使用哪个脚本？ I'm just setting the Thumbnail size : 200X200/PNG in the printer tab, under thumbnail settings.我只是在缩略图设置下设置了打印机选项卡中的缩略图大小：200x200/png。
> 
> **_Update 04-01-2025  更新04-01-2025_**
> 
> It seems that i doesn't like subfolders in the gcode folder.看来我不喜欢Gcode文件夹中的子文件夹。 letting them be in the root folder of the gcodes files folder shows the Thumbnails just fine.让它们位于GCodes文件文件夹的根文件夹中，显示缩略图很好。
> 
> can this be changed somehow in the Firmware to let it read subfolders in that folder as well? 是否可以在固件中以某种方式更改它，以便它也可以在该文件夹中读取子文件夹？[@PoohWoah](https://github.com/PoohWoah) [@bigtreetech](https://github.com/bigtreetech)

We'll take that into consideration. Thank you